bluerobotics/cockpit #1053: Widget idea/request: vehicle side-view pitch indicator

Williangalvani, 2024-07-01:
I was recently watching some videos of comercial ROVs on youtube, and one of them had a side-view  widgeteof the vehicle (that rotated)  to indicate the current pitch. I found that to be very intuitive, and I think we should do something similar.

Bonus points for having the vehicle image be configurable or come from blueos.

related to #440

ES-Alexander, 2024-07-01:
IIRC there are a couple of different interfaces that do this, and it is an intuitive way of visualising the pitch as long as the vehicle isn't too rotationally repeating from the side :-P

It could also potentially be integrated into the fancy depth indicator (#247) idea, since that already has a vehicle side-view suggested, and that could then be a compound widget showing depth+altitude+pitch in one intuitive display :-)

rafaellehmkuhl, 2024-07-01:
Cool!

@Williangalvani could you bring some screenshots/videos of those screens?