bluerobotics/cockpit #1054: On BlueOS settings sync, if there's more than one conflict, the second interaction dialog does not work

rafaellehmkuhl, 2024-07-01:
There was a first conflict for another settings variable, in which I selected to use the BlueOS value, and when I tried to do the same for this new one ("cockpit-mission-start-time"), it just does not respond.

[image]

rafaellehmkuhl, 2024-07-02:
We should wait [the interface rework PR](https://github.com/bluerobotics/cockpit/pull/1059) to be merged before working in this one, so we don't create new conflicts unnecessarily.

goasChris, 2024-07-04:
I can confirm the issue is the same for us. I personally think that it might be better to not pop multiple boxes, but just have one for any conflict, and it would resolve all.

If that is not preferable, then maybe a checkbox for "resolve all/use for all" checkbox could be added, so that a single action would resolve all the conflicts.

The mission start time will probably always be a conflict, so maybe that is something that should be ignore or have special handling.

rafaellehmkuhl, 2024-07-04:
> I can confirm the issue is the same for us. I personally think that it might be better to not pop multiple boxes, but just have one for any conflict, and it would resolve all.
>
> If that is not [...]

Good suggestion Chris! Will do that!

I'm just waiting for #1059 PR to be merged before solving this one, so we don't generate unnecessary conflicts.

PR #1081 (2024-07-08): interaction-dialog: Don't share the resolve/reject state between different instances
I love when those issues where we spend hours debugging and the solution was in one line of code. 

Happy to have solved this one. Was driving me crazy.

Fix #1054
closed 2024-07-08 by ArturoManzoli