bluerobotics/cockpit #1120: Settings sync conflicts can be more intuitive by showing the difference between the old and new settings

rafaellehmkuhl, 2024-07-16:
@ArturoManzoli has drawn a mockup for how it could look:

[image]

rafaellehmkuhl, 2025-07-24:
Closing since the solution will be to automatically solve all conflicts, as suggested on #1628.

vshie, 2025-07-25:
@rafaellehmkuhl  I think this is still worth doing! Alternative mode to whatever timestamp method, and as a debug method for the automatic resolution??

rafaellehmkuhl, 2025-07-25:
> [@rafaellehmkuhl](https://github.com/rafaellehmkuhl) I think this is still worth doing! Alternative mode to whatever timestamp method, and as a debug method for the automatic resolution??

The thing is that there's no fallback cases. The automatic resolution deals with all cases currently. There's no case in the code that leads to a conflict resolution to happen.

I think we can keep it closed and reopen it if any cases for conflict resolution appear, but I don't think this will be the case.

vshie, 2025-07-25:
I guess I'm skeptical that automatic resolution won't create situations where a change is made that wasn't desired, so if these settings were treated more like autopilot parameters that could be saved / backed up, and this interface used to see what is (selectively) applied from them, then a user has total flexibility. Timestamps can vary as you switch between computers controlling one vehicle, and different vehicles for the same computer, and many organizations have dedicated computers for this that may not be connected to the internet (ever?) or have a good time sync... so I'm wary of a invisible "final solution"

rafaellehmkuhl, 2025-07-25:
> I guess I'm skeptical that automatic resolution won't create situations where a change is made that wasn't desired, so if these settings were treated more like autopilot parameters that could be [...]

Even if a computer was not ever connected to the internet (highly unlikeable scenario), it should still have valid Date and Time settings. That's a must for almost any application nowadays (that's why it's common to receive a "you date and time appears to be off" warning if you have a computer with a faulty clock battery). I would honestly bet that a scenario where you have a computer that was never connected to the internet would probably represent a 1% population, and one that was never connected *and* have a faulty clock battery is probably 1% over this 1%.

I understand that in the long term we may want to support every single scenario out there, but for now it looks like we should stick with the 99% if we can.

My proposal: we reopen this issue, as it can be a thing for us to work in the future, moving it to a feature request, where we allow people to force a manual conflict resolution instead of accepting automatic ones. What do you think?

ES-Alexander, 2026-01-16:
@rafaellehmkuhl Please see [this comment](https://github.com/bluerobotics/cockpit/issues/1120#issuecomment-3649692158) for why I reopened it to start with - it’s a feature request now :-)