bluerobotics/cockpit #1134: Add statistics about main vehicle connection

rafaellehmkuhl, 2024-07-22:
This will help users debug problems in the connection with their vehicles.

Main one is instantaneous data rate.

It would be nice if users could see this data live.

ES-Alexander, 2025-01-31:
@rafaellehmkuhl could this be added as a colour / hover value for the existing [BaseCommIndicator](https://github.com/bluerobotics/cockpit/blob/master/src/components/mini-widgets/BaseCommIndicator.vue)?

If it was included in the data-lake then it could be easily plotted for debugging purposes :-)

rafaellehmkuhl, 2025-01-31:
> [@rafaellehmkuhl](https://github.com/rafaellehmkuhl) could this be added as a colour / hover value for the existing [...]

Definitely!
Any idea what to monitor exactly?

ES-Alexander, 2025-01-31:
> Any idea what to monitor exactly?

You already mentioned instantaneous data rate. I feel like the other main one would be latency, if that's something we're able to readily (and non-expensively) measure.

Beyond that we could potentially also track lost message proportion / inconsistent message timing (jitter?), but I feel like that might be computationally heavier than we want to deal with...

rafaellehmkuhl, 2025-01-31:
@ES-Alexander cool! Added that to this quarter as well, as it's easy and can help a lot on debugging, for both users and devs.