bluerobotics/cockpit #1139: Support Feet as a display unit in the depth widgets

rafaellehmkuhl, 2024-07-22:


patrickelectric, 2024-07-22:
I would believe that this would go towards unit internationalization as a whole over a single widget unit.
@ES-Alexander.

rafaellehmkuhl, 2024-07-22:
Yup!
Ideally we want an universal system. I thought we had an issue open for that already, but we didn't, so I just opened #1140.

In the meantime it would be good to support that on the widgets we already have.

rafaellehmkuhl, 2024-07-22:
> > Yup! Ideally we want an universal system. I thought we had an issue open for that already, but we didn't, so I just opened #1140.
> > In the meantime it would be good to support that on the widgets we already have.
> 
> I just think that supporting such thing per widget is kind of unnecessary and will take a bit of work, it may be better to just work over unit internationalization.

I would usually agree, but in this case this was an easy task (took 15 minutes to implement for both depth widgets) and as the depth is a primary state that the user tracks, it's probably a bummer for most users to not have it, preventing Cockpit usage unnecessarily.

We can take from there and actually add a global system that the widgets consume from.

PR #1141 (2024-07-24): Add support for Feet as unit of depth
[image]

https://github.com/user-attachments/assets/95710313-e0f6-47ff-8799-23c45b3af5c0

https://github.com/user-attachments/assets/7c889ae4-ff47-4013-81dc-f76444110721

Fix #1139 
Fix #1140
closed 2024-07-24 by ArturoManzoli