bluerobotics/cockpit #1146: Cockpit should be able to process generic variables into new ones ::: Z = f(X, Y)

rafaellehmkuhl, 2024-07-24:
@Williangalvani is currently on a situation where he has some (quartenion) data coming in the generic variables, and he wants to process that data into a proper angle, to be displayed in a VGI, but in the current structure, he has to create a "hardcoded" module to do that.

The idea here is for Cockpit to have a module that can be populated with user "recipes" that take data from one or more generic-variables, process them into one or more (new) variables and post them into the generic-variables stream.

Allowing the user to create those recipes in the frontend, we effectively allow for data processing without the need for hardcoded modules in Cockpit, extending it's capabilities of dealing with generic situations.

ES-Alexander, 2024-09-16:
Relevant to #613

rafaellehmkuhl, 2025-03-08:
Fixed on #1706.

closed 2025-03-08 by rafaellehmkuhl