bluerobotics/cockpit #1185: Allow user to see the performance statistics

rafaellehmkuhl, 2024-08-02:
We now have a series of performance statistics on Cockpit. We have application FPS, video player framerate and WebRTC data.
Right now they are only logged.
We should allow the user to see this data live, be it on charts, or [on overlays in Dev mode](https://github.com/bluerobotics/cockpit/pull/1179#issuecomment-2265514418).

joaoantoniocardoso, 2024-08-02:
@rafaellehmkuhl can we change to log every stats field until we decide if we don't need some of them?

rafaellehmkuhl, 2024-08-02:
> @rafaellehmkuhl can we change to log every stats field until we decide if we don't need some of them?

Yes!
Can you open a new issue for that?
I will add them before we release the charts/overlays.
Just need to test if it affects performance. If not we merge.