bluerobotics/cockpit #1201: Cockpit flatpack loaded wrong ip whitelist for video

Williangalvani, 2024-08-07:
I have seen similar issues before, where the ips are not even for existing networks.

@patrickelectric edit this

rafaellehmkuhl, 2024-08-07:
Interesting suggestion.

@joaoantoniocardoso what do you think?

Also, in case the user has a mdns as its global address, can we resolve it to match them?

joaoantoniocardoso, 2024-08-07:
> Interesting suggestion.
> 
> @joaoantoniocardoso what do you think?
I didn't get the idea of what would be the _general network ip_, and it seems tricky to do any cleanup automatically here...

> Also, in case the user has a mdns as its global address, can we resolve it to match them?

If we add an endpoint on the BlueOS's helper for that, yes