bluerobotics/cockpit #1205: minor bug: During first setup, electron postinstall checking fails

RaulTrombin, 2024-08-07:
Details:

```
➜  cockpit git:(master) uname -a
Linux rt-ThinkPad-T495 6.5.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 15 16:40:02 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

```
➜  cockpit git:(master) bun install
bun install v1.1.21 (70ca2b76)
  ⚙️  electron [1/4] /home/raulvt/Documents/BR/cockpit/node_modules/electron/install.js:45
  checksums: process.env.electron_use_remote_checksums ?? process.env.npm_config_electron_use_remote_checksums ? undefined : require('./checksums.json'),
                                                        ^

SyntaxError: Unexpected token '?'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at internal/main/run_main_module.js:17:47

error: postinstall script from "electron" exited with 1

➜  cockpit git:(master) bun install
bun install v1.1.21 (70ca2b76)

Checked 1041 installs across 1048 packages (no changes) [48.00ms]
```

rafaellehmkuhl, 2024-08-07:
How did you solve that one?

goasChris, 2024-10-09:
I would love to know as well, because I've managed to get into the same position on a clean clone:
```
bun install
bun install v1.1.30 (7996d06b)
  ⚙️  electron [1/1] Error
    at new ErrorWithFilename (/home/chris/repos/cockpit/node_modules/sumchecker/index.js:81:12)
    at new ChecksumMismatchError (/home/chris/repos/cockpit/node_modules/sumchecker/index.js:88:12)
    at <anonymous> (/home/chris/repos/cockpit/node_modules/sumchecker/index.js:61:47)
    at emit (node:events:183:48)
    at emitReadable_ (node:stream:1842:27)
    at onEofChunk (node:stream:1931:24)
    at readableAddChunk (node:stream:1868:30)
    at <anonymous> (node:stream:3248:21)
    at _flush (node:crypto:1273:13)
    at final (node:stream:3246:19)

error: postinstall script from "electron" exited with 1
```

Kernel: `Linux DESKTOP-ITM3U84 5.15.153.1-microsoft-standard-WSL2`

RaulTrombin, 2024-10-09:
> I would love to know as well, because I've managed to get into the same position on a clean clone:
>
> ```
> bun install
> bun install v1.1.30 (7996d06b)
>   ⚙️  electron [1/1] Error
>     at new [...]

Hey @goasChris,
I fixed the issue, but I wasn't able to reproduce it anymore.
Here's what I did:

Cloned the repository with the recursive option,
[Installed Node lts using nvm](https://www.geeksforgeeks.org/how-to-install-nvm-on-ubuntu-22-04/),
Installed Bun directly from project script.

Can you check your:
```bash
node --version
```
?

ArturoManzoli, 2024-10-10:
Had the same issue today, and it was basically an outdated Node version, like you guys mention, plus the vscode terminal that had an environment variable problem.

rafaellehmkuhl, 2024-10-10:
My Node is on `v21.7.1` and is working fine, if anyone needs to know where to start.

goasChris, 2024-10-11:
100% on point. It was Node related, so the error message was just very confusing, hehe.
I'm almost too embarrassed to say, but apparently I didn't even have node installed (what so ever) on this machine 🙃

PS: Maybe we should add a prerequisite in the README.md that node is a requirement, so that idiots like me won't forget to check heh.