bluerobotics/cockpit #1206: Can't execute 'bun run dev --host'

RaulTrombin, 2024-08-07:
Details:

```
➜  cockpit git:(master) bun run dev --host
$ vite --host --host
file:///home/raulvt/Documents/BR/cockpit/node_modules/vite/bin/vite.js:7
    await import('source-map-support').then((r) => r.default.install())
    ^^^^^

SyntaxError: Unexpected reserved word
    at Loader.moduleStrategy (internal/modules/esm/translators.js:133:18)
    at async link (internal/modules/esm/module_job.js:42:21)
error: script "dev" exited with code 1
➜  cockpit git:(master) npm --version
8.5.1
➜  cockpit git:(master) nodejs --version  
v12.22.9
```

RaulTrombin, 2024-08-07:
Fixed using latest nvm and nvs packages to instal node.

```
➜  ~ nvm list
       v20.16.0
->      v22.6.0
         system
default -> lts/* (-> v20.16.0)
iojs -> N/A (default)
unstable -> N/A (default)
node -> stable (-> v22.6.0) (default)
stable -> 22.6 (-> v22.6.0) (default)
lts/* -> lts/iron (-> v20.16.0)
lts/argon -> v4.9.1 (-> N/A)
lts/boron -> v6.17.1 (-> N/A)
lts/carbon -> v8.17.0 (-> N/A)
lts/dubnium -> v10.24.1 (-> N/A)
lts/erbium -> v12.22.12 (-> N/A)
lts/fermium -> v14.21.3 (-> N/A)
lts/gallium -> v16.20.2 (-> N/A)
lts/hydrogen -> v18.20.4 (-> N/A)
lts/iron -> v20.16.0

```

rafaellehmkuhl, 2024-08-07:
Nice. We should add that it will only work from a certain version of NodeJS (which we have to discover which one is).