bluerobotics/cockpit #1284: bug: widgets are not loading properly when changing views

ES-Alexander, 2024-09-02:
### Bug description

Some widgets (specifically noticed with attitude and depth widgets) do not get drawn when the active View changes.

### Steps to reproduce

1. Have an attitude or altitude telemetry value that doesn't change
2. Change views (normally or in the interface editor)
3. The relevant depth/attitude widget doesn't get drawn
    - until the telemetry value changes, or the window gets resized in a relevant direction

### Primary pain point(s)

It feels like something is broken when the expected widgets are not visible. Ideally changing view should make sure all on-screen widgets get drawn (if necessary that may require triggering every widget to get re-drawn once, although it's preferable to avoid that if possible, perhaps by just triggering a cache invalidation?).

This is reasonably low priority because in practice most vehicles will have changing telemetry values (if only because of sensor noise), and values that aren't changing are unlikely to be important to see, but it's still annoying enough that it seems worth fixing.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2024-09-02:
Damn I thought we got rid of all the canvas rendering problems already D:

ES-Alexander, 2024-09-02:
This is a pretty obscure one, and it's quite insignificant given the requirements to run into it 🤷