bluerobotics/cockpit #1319: Temperature data should have its unit configurable

rafaellehmkuhl, 2024-09-06:
The main use-case is the water temperature VGI. 
It should be easy to change between Celsius and Fahrenheit.

ES-Alexander, 2025-08-20:
Raised again and discussed somewhat in [this forum thread](https://discuss.bluerobotics.com/t/can-cockpit-display-temperatures-in-fahrenheit/21411).

ES-Alexander, 2025-08-20:
I feel like a decent way of approaching this would be to:
1. #2073
3. Create a (built in) string compound variable called `UNITS/temperature` = `{{ UNITS/displayPreference }} == "metric" ? ℃ : ℉`
4. Create the following numerical compound variables:
    - `Temperature Celsius` = `{{ SCALED_PRESSURE2/temperature }} / 100`
    - `Temperature Fahrenheit` = `{{ temperature-celsius }} * 1.8 + 32`
    - `Temperature` = `{{ UNITS/displayPreference }} == "metric" ? {{ temperature-celsius }} : {{ temperature-fahrenheit }}`
5. Allow using string data-lake values in the units of VeryGenericIndicator widgets
6. Re-implement the water temperature preset as using `{{ temperature }}` and `{{ UNITS/temperature }}`