bluerobotics/cockpit #1326: frontend: Add GPS default for widget and allow changing significant digits displayed

ocean-engineer, 2024-09-09:
### Current behaviour

The interface seems to have a fixed number of significant digits so trying to display latitude/longitude as a widget or as an overlay on a recorded video cuts it off to at most the first 6 digits or if you add the multiplier to get from the integer position to decimal degrees only 4 digits ie. 34.01 degrees. Images show with and without a multiplier.
[image]
[image]
[image]

[image]

### Expected or desired behaviour

The ability to change the number of digits displayed for any of the variables displayed as widgets or overlays. Additionally adding a default widget for GPS position could include the multiplier to go from integer position to decimal degrees. I need to display more digits with the latitude/longitude data.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ocean-engineer, 2024-09-11:
Relevant to #992. #606 seems to be what disallowed changing the digits.

rafaellehmkuhl, 2024-09-11:
Yup. It was an intentional change that ended up having this unintentional side effect.

Our goal was to make this widget as simple as possible to use and at the same time making it look nice in any situation. I think we archived that but created this problem, and the coordinates case is probably the main one that does not fit the currently implementation.

Probably the least we can do to solve this is to add optional parameters to the VGI that would allow the user to set a custom width and also a custom number of digits for the value. What do you think about it?

ocean-engineer, 2024-09-11:
Totally understand, I can definitely see why you changed it. I really like the simplicity of the interface, but yeah having a way to customize it for cases like this seems necessary in some way.

I think optional parameters for width and number of digits seems ideal - most people shouldn't need to touch them, but it lets you display longer variables if needed. Eventually adding presets for the Lat and Long would also avoid the need for most people to mess with them.

patrickelectric, 2024-09-26:
That was raised again during the SW-SMS meeting

rafaellehmkuhl, 2024-09-26:
I have a friend that started working at this issue last week. I will talk to him to see how things are going.

rafaellehmkuhl, 2024-11-04:
This was fixed by @ArturoManzoli on #592.

closed 2024-11-04 by rafaellehmkuhl