bluerobotics/cockpit #1340: mini-widget: display a location

ES-Alexander, 2024-09-16:
### Problem to solve

Sometimes it's useful to track/record the numerical values of a location, rather than just visualising its position on a map, and sometimes locations are not geo-referenced. Currently this is awkward to do, because it requires multiple separate mini-widgets and the data may not be in the desired display format.

### Desired approach

A mini-widget with support for 
1. any valid sources of location data
    - MAVLink / NMEA / etc
    - vehicle location, control station location, etc
2. common display formats
    - latitude, longitude
    - UTM
    - local x, y [,z] (probably with support for metric vs imperial)
3. position validity/accuracy
    - colour or opacity could potentially be used to indicate when a position estimate is uncertain, or older than a threshold time

### Additional context

- [Lat, Lng](https://github.com/bluerobotics/cockpit/issues/992) as part of a broader VGI discussion
- [UTM](https://discuss.bluerobotics.com/t/utm-format-position-display-in-cockpit-qgc-for-recording/18008) request and discussion

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2024-09-16:
I would add GPS position estimation as well, like BlueOS GPS widget.

vshie, 2024-09-16:
Having a button to mark/save a given position, give it a label, could be super handy for both ROV and Boat...

vshie, 2024-09-16:
Ah, good point, my apologies!

ES-Alexander, 2024-09-16:
No worries - just don't want your good idea to get lost in the weeds! :-)