bluerobotics/cockpit #1343: bug: Drag button of edit mode panel is not working

patrickelectric, 2024-09-17:
### Bug description

You can click and hold to drag, but it's not working 

[image]

### Steps to reproduce

Go to edit interface mode, center edge of edit panel, can't drag.

### Primary pain point(s)

Does not work

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2024-09-19:
I'm not sure if I got the problem. What are you trying to drag?
Only the regular widget entries are draggable in this menu.

patrickelectric, 2024-09-19:
If you hover the mouse over the edge of the menu you can see that the drag button shows up to resize it.

rafaellehmkuhl, 2024-09-19:
> If you hover the mouse over the edge of the menu you can see that the drag button shows up to resize it.

I'm not seeing it here.
Are you sure you don't have a widget fullscreened and this is what is being resized?

patrickelectric, 2024-09-19:
It's indeed the fullscreen widget!

closed 2024-09-19 by rafaellehmkuhl