bluerobotics/cockpit #1352: bug: Sliding the 'Arming' Button makes the on screen text to be selected

ArturoManzoli, 2024-09-20:
### Bug description

Even though the 'Arm' button has to be pressed to arm, it has a shape that invite the user to slide it. If the user does so, the text present on the screen get selected and its hard to unselect,

### Steps to reproduce

1- Change to Map View
2- Slide the 'Disarmed' button to the right

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

vshie, 2024-10-28:
I would be epic if, like QGC, you could also hold the Space bar to arm / disarm. Does this idea need to be in its own issue?

rafaellehmkuhl, 2024-10-29:
> I would be epic if, like QGC, you could also hold the Space bar to arm / disarm. Does this idea need to be in its own issue?

Hey Tony! Yes! Please open a new issue for that one. And actually the base code for that is pretty much done, so it would be a hard one to implement.

closed 2024-10-29 by rafaellehmkuhl