bluerobotics/cockpit #1367: frontend: signal strength indicators

ES-Alexander, 2024-09-26:
### Current behaviour

Current behaviour is binary - just connected or disconnected, which means operators don't have a good sense of when the connection is poor / at risk of being lost until it's too late.

### Expected or desired behaviour

Show signal strength indicators for the connected vehicle, and, if present, a base station.

Per [this forum post](https://discuss.bluerobotics.com/t/suggestion-for-signal-strength-in-qgroundcontrol/18106) QGC apparently has a signal strength indicator, albeit without splitting out the connections strength to the vehicle.

Per #1099, satellite connectivity for position estimates should also be available, like it is in BlueOS.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

vshie, 2024-09-26:
For the BlueBoat using Mikrotik radios, BlueOS should be able to request the signal strength and more usefully, the signal to noise ration (SNR) of the WiFi connection. This will always report from 192.168.2.4 (or whatever the onboard radio device IP is.) AI search suggests librouteros (python) can be used to communicate and retrieve this. 
Potentially hallucinated code: 
`   import routeros_api

   connection = routeros_api.RouterOsApiPool('YOUR_ROUTER_IP', username='YOUR_USERNAME', password='YOUR_PASSWORD')
   api = connection.get_api()`
`   # Get interface information
   interfaces = api.get_resource('/interface/wireless')
   for interface in interfaces.get():
       if 'snr' in interface:
           print(f"Interface: {interface['name']}, SNR: {interface['snr']}")`

vshie, 2024-09-26:
Also, as SNR goes below 20, signal loss is imminent and usable bandwidth minimal.... Suggest this should be 1 bar / no bar indication, with above 45 full signal strength.

rafaellehmkuhl, 2024-09-26:
Great idea! We just need to add this API in BlueOS first.

Also, we should have default values but allow configuration of the warning and critical thresholds.