bluerobotics/cockpit #1412: bug: iframe removed but still shows up, until edit interface opened

vshie, 2024-10-16:
### Bug description

iframe was configured to hold DVL screen. Interface widgets added. Iframe removed. Cockpit closed, re-opened, iframe still present. When edit interface is opened, iframe immediately dissapears. 

### Steps to reproduce

Add iframe widget. Add other generic indicator widget. Remove iframe. Close cockpit. Reopen cockpit, observe iframe still present. Open edit interface, observe iframe dissapears. 

### Primary pain point(s)

Changes not sticking!

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2024-10-17:
I've tried to reproduce the bug here many times without success. 
Is that something that is happening consistently for you? If so, can you record the screen so I can try to find out what am I doing differently?

vshie, 2024-10-17:
definitely consistent, observed it occur 4 times in a row. Wasn't on my computer, so will record first thing during tomorrows tests! Can get cockpit logs to you as well, anything else helpful?

rafaellehmkuhl, 2024-10-17:
> definitely consistent, observed it occur 4 times in a row. Wasn't on my computer, so will record first thing during tomorrows tests! Can get cockpit logs to you as well, anything else helpful?

Just the video is fine! The logs won't probably help much in this case.