bluerobotics/cockpit #1415: we should allow "conditional" mini-widgets

Williangalvani, 2024-10-17:
some data is only really important if out-of-bounds
Cpu usage and temperature, for example

rafaellehmkuhl, 2024-10-26:
The idea here is to only show the widget if the condition happens, or to allow changing its color in this case, for example?

Williangalvani, 2024-10-26:
> The idea here is to only show the widget if the condition happens, or to allow changing its color in this case, for example?

Yes. As we get more and more widgets, I think it would be easier for the pilot to "navigate" the UI if information that is not immediately important was hidden.

we could for example, only display the temperature mini-widget after it goes above 75 Celsisus.
Same could be done for cpu clock, only shows something if it is throttled

rafaellehmkuhl, 2024-10-26:
That could actually look really interesting if there was an animation on appear and disappear.