bluerobotics/cockpit #1424: widget: 2D path plan over map

Ryanf55, 2024-10-27:
### Problem to solve

I want to visualize my path plan in the map view. 

### Desired approach

Plot the path plan (set of waypoints). Show waypoint index number on hover and the lines connecting the points

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2024-10-28:
Hi @Ryanf55,

Just to clarify, are you suggesting adding an overlay to the map widget, to visualise the mission currently on the vehicle / being executed?

> Show waypoint index number on hover

While I think showing metadata (e.g. showing commands applied to a given waypoint) as icons and/or on hover makes sense, I'm curious what value the index number specifically would provide? From a quick review of [the MAVLink mission protocol](https://mavlink.io/en/services/mission.html), the only things the ID are relevant to are setting a new current mission item (target waypoint) and tracking which waypoints have already been reached, both of which seem more intuitive to display via other aspects of the UI (e.g. the colours and shapes of the waypoint icons).

Ryanf55, 2024-10-29:
Yes, I would like a way to view the path the vehicle is going to take. For starters, this could be the mavlink defined mission. Later, it would be nice for a companion computer to send arbitrary paths to the UI to display that were computed from a path planner. 

There's some examples here (3D and 2D views)
https://discuss.ardupilot.org/t/offboard-terrain-navigation-for-plane-with-ros-2-ap-dds/114418

Ryanf55, 2024-11-16:
I've been doing some work in 3D with [MAVCesium](https://github.com/ardupilot/MAVCesium). It supports 2D and 3D. 
MAVCesium is currently tied to mavproxy, but a lot of the underlying javascript could be made more generic. 

If I adapted MAVCesium to work independently of mavproxy, made it interact with the vehicle only through the MAVLink JSON interface, would it be possible to use it inside of Cockpit?