bluerobotics/cockpit #1428: frontend: Increase maximum zoom level on map to allow for route planning

vshie, 2024-10-28:
### Current behaviour

This is the max zoom level: 
[image]

It would be nice if you could get this close: 
[image]

### Expected or desired behaviour

Allow maximum zoom level that tile provider supports

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Michael-F-Bryan, 2024-11-07:
This would be really nice for things like jetty testing.

rafaellehmkuhl, 2024-11-07:
Just to add on that, what is currently limiting our zoom level are the map tiles. We are using ESRI tiles, which have a maximum zoom level of 19. Other providers like Google allow for a closer zoom, but their terms of service disallow this type of usage.

If anyone knows of an alternative tile provider that allows for closer zoom levels, please let us know.

Below is a comparison between ESRI and Google on max zoom:

Google:
[image]

ESRI:
[image]

closed 2025-04-24 by ES-Alexander