bluerobotics/cockpit #1449: Issue configuring one joystick when there are multiple joysticks present

Williangalvani, 2024-11-13:
Eldin had two joystick detected. one was actually a 3d mouse (emulator), the other was our regular xbox360.
Visual gamepad mapper sometimes shows up (but wouldn't work/ highlight buttons), sometimes goes straight to the table view.
Editing something in the table view managed to bring back the visual editing interface, and then everything worked as expected.

curtnichols, 2025-05-05:
Our application uses a selected joystick for control of our hardware; when Cockpit is also running using our joystick appears to affect the Cockpit software. Does Cockpit have a means of selecting a specific joystick?

rafaellehmkuhl, 2025-05-05:
Hi @curtnichols!

Cockpit does not have this option yet, indeed, although it's something that I guess wouldn't take long to add, and we are currently working on joystick-related issues, so the timing is good.

Can you get more details on exactly how Cockpit seems to being affected on your customer's setup? From the current implementation, I would expect the system to be receiving commands from both joysticks at the same time, which makes the sticks basically unfunctional.

curtnichols, 2025-05-05:
@rafaellehmkuhl Your description ("receiving commands from both joysticks at the same time") is in line with the info I received. Both joysticks appear to be driving Cockpit. Thanks for the info.

rafaellehmkuhl, 2025-05-05:
> [@rafaellehmkuhl](https://github.com/rafaellehmkuhl) Your description ("receiving commands from both joysticks at the same time") is in line with the info I received. Both joysticks appear to be driving Cockpit. Thanks for the info.

Cool! I will add the option to choose a single joystick to be the one controlling Cockpit.

rafaellehmkuhl, 2025-05-06:
@curtnichols I have implemented the solution on #1874. It needs some of the changes made in #1854 to be merged before, so I expect to have both merged by next week.