bluerobotics/cockpit #1450: frontend: Request to add page that contains sensor callibration and failsafes

hafizhkarim, 2024-11-15:
### Current behaviour

there are no settings about vehicle sensors and failsafes

### Expected or desired behaviour

added an extra page for sensor calibration and vehicle failsafe

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2024-11-20:
Hi @hafizhkarim!

Currently we are focused in building on Cockpit only functionalities that we consider to be specifically of the ground control station. 

For the moment we are relying on BlueOS to implement functionalities related to vehicle setup, like sensor configuration and failsafe settings.

This may change in the future, but for the moment we rely on external contributions to have this on Cockpit sooner.

ES-Alexander, 2025-02-07:
This has also been raised and closed previously.

#1051 is a potential solution/workaround while we're ok with Cockpit being reliant on BlueOS.

closed 2025-02-07 by ES-Alexander