bluerobotics/cockpit #1489: frontend: Improve Boat's default controller mappings

ArturoManzoli, 2024-12-05:
### Current behaviour

While controlling a boat, a lot of important function assignments are missing.

### Expected or desired behaviour

We should add as default, a good set up for xbox and dualsense controllers.
As a stretch, good support for flight-sticks would be very nice too.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2024-12-05:
Do you remember exactly which functions did you guys miss? I remember talking to William at some point and he said that the only ones that we were missing where they arm and disarm that we now have.

ArturoManzoli, 2024-12-05:
> Do you remember exactly which functions did you guys miss? I remember talking to William at some point and he said that the only ones that we were missing where they arm and disarm that we now have.

We did miss mainly the mode selector, and that's what I started the issue for. But checking on the default joy profile, its clear that we can add many other functionalities for surface vehicles. There are a lot of empty keys to use

rafaellehmkuhl, 2024-12-06:
Related to #348.