bluerobotics/cockpit #1498: frontend: editor: move mini-widget bar to mini-widget section

ES-Alexander, 2024-12-10:
### Current behaviour

Adding extra mini-widgets bars is done via the regular widget option in the editor, which is somewhat unintuitive to find (a container likely isn't what most people think of as a widget).

### Expected or desired behaviour

It would likely make more sense to add a dedicated button for it below the dropdown when in the mini-widget section, like is being done with the "add widget base" button in the input widgets section.

I'm not sure whether we'd want that to be instead of the regular widget approach, or just support both 🤷 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2024-12-10:
Totally agree. It makes sense to be the same button as the "add a widget base".

Also, we should understand how to deal with the existence of those two (mini-widget bar and custom widget base). They seem to be doing the same thing, although in different ways.

I understand that the mini-widget bar is doing a better job in the widget organization (see image below) while the custom widget base has more functionality (import/export layouts and collapsing).

[image]

I think we should merge the two.

ES-Alexander, 2024-12-10:
> I think we should merge the two.

Should input widgets just be a type of mini-widget, instead of being a separate thing?

Not sure if we’d still want separate selectors, or just some kind of icon to show whether one accepts interactions/data or displays stuff.