bluerobotics/cockpit #1501: bug: video that is flipped or rotated in Cockpit does not record / save in this adjusted orientation.

vshie, 2024-12-10:
### Bug description

Video streams can (very nicely) be rotated (even multiple times1) or flipped. However, when the recording is downloaded, the video is in the original orientation. 

### Steps to reproduce

Rotate video stream in edit interface menu, then make a recording. Rotation/flip of displayed video will not match recording. 

### Primary pain point(s)

The browser decodes, and then we flip, but João mentions that it may be possible to hint the correct orientation in the recording? 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-12-11:
Yeah, the video's metadata contains a `rotation` field for that purpose. Cockpit should be able to change it after processing the chunks.

rafaellehmkuhl, 2024-12-11:
> Yeah, the video's metadata contains a `rotation` field for that purpose. Cockpit should be able to change it after processing the chunks.

Problem is we have to reprocess the chunk, right?

joaoantoniocardoso, 2024-12-11:
> > Yeah, the video's metadata contains a `rotation` field for that purpose. Cockpit should be able to change it after processing the chunks.
> 
> Problem is we have to reprocess the chunk, right?

So the two options are:
1. Change the metadata of the final video .mp4 file and don't mess with the file data, should be quick.
2. Decode, change rotate/flip frame frame, then re-encode every frame. We might want to do it since we are going to have a video processing pipeline anyway (for color lens and color correction, for example)

ES-Alexander, 2024-12-12:
Option 2 seems useful, but advanced. In cases where there isn't already processing happening it would be preferable to support option 1, especially for the sake of lower end systems that could struggle with encoding.

rafaellehmkuhl, 2024-12-12:
Although option 1 seems ideal, I'm struggling to find any way to do this in the browser @joaoantoniocardoso. If you have any clue let me know.

Would be [ffmpeg.wasm to](https://github.com/ffmpegwasm/ffmpeg.wasm) go.

joaoantoniocardoso, 2024-12-12:
> Would be [ffmpeg.wasm to](https://github.com/ffmpegwasm/ffmpeg.wasm) go.

Yep, the ffmpeg can do that, I expect the wasm version to also be capable. Maybe something like https://www.npmjs.com/mp4js could also do that

ES-Alexander, 2024-12-30:
Noting that semantically the video recorder mini-widget acts on the incoming video stream, and is unrelated to any video display widget configuration options. Multiple display widgets could show the same stream in different ways, and we generally don't want or expect things like the widget scale or cropping to affect the recorded video.

If we wanted to record a stream as flipped/rotated then we'd need to add controls for that in the video configuration section, and potentially remove them from the video widget (if we don't expect users to need to handle them independently for different outputs).