bluerobotics/cockpit #1550: bug: mission planning: dragging and removing waypoints doesn't work

ES-Alexander, 2025-01-08:
### Bug description

1. Dragging mission waypoints drags a circle but not the point label or the lines connecting it to the rest of the path
2. Points are labelled with 1-indexing on the map, and 0-indexing in the list on the right side
3. Deleting a point deletes the label and list entry, but keeps the line
4. Waypoint labels also seem to have huge padding for no reason
[image]

### Steps to reproduce

1. Open Mission Planning mode
2. Add some mission waypoints
3. Try to drag them
4. Try to delete them

### Primary pain point(s)

Mission planning is much harder if the waypoints aren't editable, and if the labels are so big they have significant overlap.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2025-01-11:
Same issue here

rafaellehmkuhl, 2025-01-28:
Can confirm. Will be in our quarter milestone.

ArturoManzoli, 2025-01-30:
> ### Bug description
> 1. Dragging mission waypoints drags a circle but not the point label or the lines connecting it to the rest of the path
> 2. Points are labelled with 1-indexing on the map, and [...]

1- Solved in the first mission planning refactor [PR-1469](https://github.com/bluerobotics/cockpit/pull/1469);
2- The right side waypoints panel was replaced by a sliding configuration panel, as the one present on the Edit Mode to configure input widgets - [PR-1523](https://github.com/bluerobotics/cockpit/pull/1523);
3- Also fixed on - [PR-1523](https://github.com/bluerobotics/cockpit/pull/1523);

4- I suspect this issue is related to a lifecycle problem. It loads the correct CSS properties only after a page refresh. My suggestion is to temporarily add an auto refresh until we find the root cause of the problem.

With the auto refresh logic added (also showing the issues from 1 to 3 solved):

https://github.com/user-attachments/assets/77bbeec3-bd9e-4a08-9bdd-fb368b8b3a1f

rafaellehmkuhl, 2025-01-30:
> 1- Solved in the first mission planning refactor [PR-1469](https://github.com/bluerobotics/cockpit/pull/1469);
> 2- The right side waypoints panel was replaced by a sliding configuration panel, as [...]

I've seen problems 1 and 3 yet happening after we merged #1523. It is intermittent. Happens on some boots, not always.

PR #1639 (2025-01-31): Prevent CSS leakage into mission planning module
As discussed on the issue #1550, all the bugs described there were fixed in previous pull requests related to mission planning (#1469 and #1523), but the item 4.

@JoaoMario109 pointed out a possible CSS leakage from the Map component, that indeed was the cause of the styles only being applied correctly after a page refresh. Kudos to him

Closes #1550
closed 2025-01-31 by ArturoManzoli