bluerobotics/cockpit #1551: bug: electron: macOS vehicle discovery not working

ES-Alexander, 2025-01-08:
### Bug description

Searching for vehicles seemingly results in an error traceback, which is way too big to fit in the card:
[image]

### Steps to reproduce

1. Install standalone Cockpit on macOS (Intel, Sequoia 15.1.1)
2. Open it
3. Try to search for vehicles
4. It fails

### Primary pain point(s)

Requires manual vehicle connection, which can be annoying.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-01-08:
Humnnn, maybe for some reason it's only working on Apple silicon? Strange, there shouldn't be anything in the code specifically to it.

Could you look at the element and the console and bring the full error code?

ES-Alexander, 2025-01-10:
> Could you look at the element

This is the very long element:
```html
<span class="v-btn__content" data-no-activator="">&lt;!--
Copyright : (C) 2014 Sagemcom - URD2

This software and source file is the property of Sagemcom
and may not be copied or used without prior written consent.
--&gt;
&lt;!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd"&gt;
&lt;html xmlns="http://www.w3.org/1999/xhtml" class="no-js"&gt;

&lt;head&gt;
	&lt;meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1" /&gt;
	&lt;meta charset="utf-8" /&gt;
	&lt;meta name="sagemcom" content="oui" /&gt;
	&lt;meta name="viewport" content="width=device-width,height=device-height,initial-scale=1.0,user-scalable=no,user-scalable=0" /&gt;
	&lt;meta http-equiv="content-language" content="en"&gt;
	&lt;base href="" id="baseHref" /&gt;
	&lt;script&gt;
		var initialRegex = /(.*)\/.*.html/i;
		var baseHref = initialRegex.exec(window.location.pathname);
		var usethis = window.location.pathname;
		if (baseHref) {
			usethis = baseHref[1] + '/';
		}
		document.getElementById("baseHref").href = usethis;
	&lt;/script&gt;
	&lt;style&gt;
		meta.foundation-data-attribute-namespace {
			font-family: false;
		}
	&lt;/style&gt;

	&lt;link id="favicon" rel="shortcut icon" href="images/faviconNone.ico" /&gt;

	&lt;title&gt;&lt;/title&gt;

	&lt;script&gt;
		window.onerror = function (msg, url, line, col, error) {
			if (typeof SboxInterface !== 'undefined') {
				SboxInterface.RemoveLoader();
				setTimeout(function () {
					if (document.getElementById('main-ui-view').innerHTML.length === 0) {
						SboxInterface.processHTML('retry');
					}
				}, 1000);
			}
			return false;
		};
	&lt;/script&gt;

	&lt;script src="js/vendor.js?9ee5fed1ec35c7b9"&gt;&lt;/script&gt;

	&lt;script&gt;
		// fix for old code in gui-core relying on $.browser
		if (typeof (jQuery) != 'undefined') {
			jQuery.browser = {};
			jQuery.browser.msie = false;
			jQuery.browser.version = 0;
			if (navigator.userAgent.match(/MSIE ([0-9]+)\./)) {
				jQuery.browser.msie = true;
				jQuery.browser.version = RegExp.$1;
			} else if (!!navigator.userAgent.match(/Trident\/7\./)){
				jQuery.browser.msie = true;
				jQuery.browser.version = 11;
			}
		} else {
			jQuery = {};
			jQuery.browser = {};
			jQuery.browser.msie = true;
			jQuery.browser.version = 8;
		}
		svg4everybody();
	&lt;/script&gt;
	&lt;script src="js/libs.js?88395d9c637cb1a0"&gt;&lt;/script&gt;

	&lt;script src="js/jquery-utils.js"&gt;&lt;/script&gt;
	&lt;script src="js/gui-core.js"&gt;&lt;/script&gt;
	&lt;script src="js/gui-api.js"&gt;&lt;/script&gt;
	&lt;script src="js/gui-widgets.js"&gt;&lt;/script&gt;

	&lt;script src="js/scripts.js?bfd765d7cbca0e7d"&gt;&lt;/script&gt;

	&lt;script src="js/config.js?da710c25cdb4cdf3"&gt;&lt;/script&gt;

	&lt;link rel="stylesheet" href="styles/vendor.css"&gt;
	&lt;link rel="stylesheet" href="styles/vendor-rtl.css"/&gt;

	&lt;link rel="stylesheet" href="styles/lib.css"&gt;

	&lt;!--[if IE 8]&gt;
	&lt;link rel="stylesheet" href="../styles/foundation-ie8.css" /&gt;
	&lt;![endif]--&gt;

	&lt;link rel="stylesheet" href="styles/main.css"&gt;
	&lt;link rel="stylesheet" href="styles/main-rtl.css"/&gt;

	&lt;link rel="stylesheet" href="styles/dev.css"&gt;
	&lt;link rel="stylesheet" href="styles/dev-rtl.css"/&gt;
&lt;/head&gt;

&lt;body ng-app="sboxApp" ng-class="isMainTemplate ? '' : 'bg-gradient'" class="bg-gradient-force"&gt;
	&lt;div wait&gt;&lt;/div&gt;
	&lt;div id="main-ui-view" class="zoomViewport" ui-view&gt;&lt;/div&gt;
&lt;/body&gt;

&lt;/html&gt;
 <span class="text-xs ml-2 opacity-50">(10.0.0.138)</span></span>
```

It seems to find one or two others inconsistently, as:

- BlueROV2 (192.168.2.2)
- BlueROV2 (10.0.0.21)

So I guess it's working largely as intended, but is sometimes picking up an extra thing that gives an unexpected response as the vehicle type / name?

> ... and the console and bring the full error code?

The console doesn't have any special errors raised while searching (just the normal connection failures):

[image]

rafaellehmkuhl, 2025-01-13:
I have just tested here and on mine it also found the available vehicles and is showing the incorrect element. Will investigate.

rafaellehmkuhl, 2025-01-21:
Interesting: it works fine in the development version, with the bug only happening in the production build.

[image]

PR #1605 (2025-01-22): Do not show devices other than BlueOS vehicles in the vehicle discovery
Turns out some devices like routers return 404 pages with status code 200 if you try to access inexistent subdomains of them.

Fix #1551
closed 2025-01-22 by rafaellehmkuhl