bluerobotics/cockpit #1553: bug: telemetry data is over wizard

patrickelectric, 2025-01-08:
### Bug description

[image]

### Steps to reproduce

1. Run wizard
2. Click next until telemetry window is over wizard

### Primary pain point(s)

Can't read wizard, the wizard also does not come up the telemetry window if clicked over it

### Additional context

bad initial experience 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-01-08:
@ArturoManzoli can you check if your PR deals with that?

ArturoManzoli, 2025-01-09:
> @ArturoManzoli can you check if your PR deals with that

The PR doesn't deal with that situation. Actually, this layering was thought on the Tutorial feature implementation. 

If you click on the tutorial screen (just like in any other window management system) it will bring up the selected window, and the tutorial will be over the telemetry screen again and vice-versa.

Also in very small screens (not our attention priority for now), the layers order gets very problematic and the user needs to close the telemetry window to access back the tutorial.

rafaellehmkuhl, 2025-01-13:
> The PR doesn't deal with that situation. Actually, this layering was thought on the Tutorial feature implementation.
>
> If you click on the tutorial screen (just like in any other window management [...]

Humnnnn, totally agree. And I just checked and clicking the tutorial window does indeed bring it to the top. @patrickelectric 
 you mentioned the opposite. Can you confirm that? Otherwise we can close the issue.

https://github.com/user-attachments/assets/143bdd1d-920b-4650-8ee3-5a1ecc7021a2

patrickelectric, 2025-01-24:
From a fresh intall of 1.9.0

[image]