bluerobotics/cockpit #1564: [mission planning] performance with huge missions

joaoantoniocardoso, 2025-01-13:
I think we need to warn the user when the area is too big, otherwise this can happen without notice:

[image]

rafaellehmkuhl, 2025-01-13:
Oh god 🤣 

I think we can have a configuration somewhere with `maxNumberOfSurveyPoints`, limit it, and if the survey reaches it we tell the user it was not able to create because of that limitation and if they want they can raise the limit before trying again.

What do you think?

joaoantoniocardoso, 2025-01-13:
sounds reasonable =)

ES-Alexander, 2025-08-06:
Not a direct warning, but #1459 will likely help prevent this - you're less likely to try to generate such a big survey if you know how large the area is :-)