bluerobotics/cockpit #1591: JavaScript code in the JS actions and DIY widget should not accumulate on save

rafaellehmkuhl, 2025-01-17:
Today, for both, when te user saves the code, if there are leftovers like `setInterval`s or `setTimeout` or running loops, they will stay running and the new code will be instantiated, so there can be conflicts (like two intervals trying to change the color of the same component).

ES-Alexander, 2025-01-20:
Is it possible to automatically determine which interval/timeout IDs need to be cleared, or do we need to recommend/require that programs using such patterns should provide a cleanup function or something?

rafaellehmkuhl, 2025-01-20:
> Is it possible to automatically determine which interval/timeout IDs need to be cleared, or do we need to recommend/require that programs using such patterns should provide a cleanup function or something?

We can ensure if we add a custom "window.cockpit.setInterval(...)" function and keep track of the IDs, otherwise I'm not sure.
Lago suggested something. Will talk to him.

closed 2026-09-28 by rafaellehmkuhl