bluerobotics/cockpit #1592: DIY widget should allow easy import/export

rafaellehmkuhl, 2025-01-17:
This way people can share their creations easily.

ES-Alexander, 2025-01-20:
Is this feasible as a single HTML file, with a `style` and `script` section? :-)

rafaellehmkuhl, 2025-01-20:
> Is this feasible as a single HTML file, with a `style` and `script` section? :-)

It is! The problem is if the user adds style/script tags to the html tag 😂 But we can treat this case as well.

Your use case is being able to use/test it outside cockpit right?

ES-Alexander, 2025-01-20:
I didn’t have a particular use-case in mind - it just seemed like a good fit for the job, since html files are already able to contain css and JavaScript, and then each DIY widget could be confined to (and shared as) a single file :-)

PR #1730 (2025-03-10): DIY widget: Allow exporting and importing configuration
Fix #1592
closed 2025-03-10 by rafaellehmkuhl