bluerobotics/cockpit #1597: DIY widget displays as having no configuration option

ES-Alexander, 2025-01-20:
### Bug description

The cog shows as disabled, even though clicking it opens up valid configuration:
[image]

### Steps to reproduce

1. Open Edit Mode
2. Add a DIY Widget
3. The configuration cog for it (in the "Widgets in View" list) shows up as disabled 😢
4. Clicking the seemingly disabled cog still actually works to open the configuration 🥲

### Primary pain point(s)

Critical control options should not be hidden behind a button that shows itself as un-clickable 😅 

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-01-20:
Will fix that now.

rafaellehmkuhl, 2025-01-20:
Strange that my local typescript linter is blaming on the config list for not having a value for the DIY widget, but the CI is not.

PR #1598 (2025-01-20): widgets: Enable config menu for the DIY widget
[image]

Fix #1597
closed 2025-01-20 by patrickelectric