bluerobotics/cockpit #1651: frontend: Add language switching

zshun-robot, 2025-02-04:
### Current behaviour

The current interface is only available in English, but I want a Chinese Version.

### Expected or desired behaviour

Is it possible to add a language switch function to the settings? For example, add a Chinese and English switch.

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2025-02-04:
Hi @zshun-robot, this Issue is a duplicate of #409, so I'm closing it to keep discussion consolidated there. You are welcome to share any relevant thoughts in that Issue, and/or contribute a Pull Request to resolve it :-)

closed 2025-02-04 by ES-Alexander

rafaellehmkuhl, 2025-02-04:
Thanks @ES-Alexander.

@zshun-robot we would love to support multiple languages, but right now we are out of hands to put on that :(

As Eliot said thought, PRs around that would be more than welcomed. Feel free to participate on our biweekly meetings or reach us directly to talk about that if you want to know where to start.

ES-Alexander, 2026-01-08:
> I want a Chinese Version.

Noting that #2314 is working on adding this :-)