bluerobotics/cockpit #1683: DepthHUD working as intended? Adding a surface GPS introduces strange behaviour

samuel-reedy, 2025-02-10:
I’m seeking clarification on the behaviour of the Depth HUD and Depth Mini-widget after we’ve noticed a discrepancy when attaching a surface GPS to the BlueROV2. 

- Without the GPS both Depth HUD and Depth Mini-widget read 0 at the surface.
- With a surface GPS attached the DepthHUD appears to show 0 on the graticule but shows a large negative number on the surface (see screencapture). While Depth Mini-widget still shows 0. 

Closer examination of the code for the [DepthHUD](https://github.com/bluerobotics/cockpit/blob/master/src/components/widgets/DepthHUD.vue) and [Depth Mini-Widget](https://github.com/bluerobotics/cockpit/blob/master/src/components/mini-widgets/DepthIndicator.vue) shows both using the same value from altitude.msl but treating these differently:

- For the Depth Mini-Widget, the depth is set to 0 for values before 0.01. Meaning in a body of water above sea level, the depth always gets set to 0.
- Whereas the Depth HUD does not have this same limitation and always displays the depth, even if the ROV is far above sea level. 

Can you clarify what the intention is here? Our assumption is that the depthHUD (due to its name) would always show depth below a waterline. So, whether that’s operating in the sea at sea level or up in a lake at an altitude above sea level, it would show depth below the waterline. Is that the correct intention?

Wouldn’t Relative Altitude (e.g. GLOBAL _POSITION_INT.relative_alt) be a better option? This is the [altitude above Home](https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT). If we interpret Home as being on the surface of the water (i.e if set at start of a mission), isn’t that exactly what’s needed? Would that not better meet the need for displaying depth irrespective of the altitude of the body of water you’re deployed in?

We found [this Ardupilot documentation on interpretation of altitude](https://ardupilot.org/copter/docs/common-understanding-altitude.html) but it doesn’t include the Sub use case.  I’m assuming that Cockpit is currently a Sub-only implementation but with eye to make it more general purpose for drone use, right? In which case a separate AltitudeHUD widget would be required? 

Keen to get to the bottom of this as trying to get the surface GPS + DVL solution working smoothly - showing and logging correct depth values. So, understanding the direction here would be super useful.

[image]

rafaellehmkuhl, 2025-02-13:
Hey @samuel-reedy! Sorry for the delayed response.

You're correct on assuming it should only show values below 0, yes.

Regarding using `GLOBAL_POSITION_INT.relative_alt` instead of `AHRS2.altitude`, it really depends on the situation. For me right now (~800m above sea level), the first shows a value around -200 m while the second gives me around 0 m.

We are currently using `AHRS2.altitude` there because it is what sub users usually use, indeed, because it zeroes on boot and is populated by a water pressure sensor (when available), so you will generally have a good value for the water depth.

All that said, let me tell you about our plans: some months ago we introduced a new system on Cockpit called DataLake, where all sorts of variables are stored. What we are in the process of doing is to migrate all the widgets to this system, so people will be able to select which source of information they want to have for each widget. That means you will be able to decide if you want the `DepthHUD` to show `AHRS2.altitude` or if you prefer to use `GLOBAL_POSITION_INT.relative_alt`, or maybe even populate the widget with an external sensor that is not integrated to Ardupilot (you can do that through a JavaScript action + a DataLake variable + a `VeryGenericIndicator` widget today).

samuel-reedy, 2025-02-17:
@rafaellehmkuhl Thanks for the response.

We're seeing incorrect depth values. The mini widget is always showing 0m, and the DepthHUD is showing large negative values. Of note is that the depth HUD widget does move with what appears to be the correct change in depth, but just with the large offset. Does this mean we've got something setup incorrectly if we should be seeing the water pressure sensor reading? 

Cheers