bluerobotics/cockpit #1699: Actions overview should display if they are linked to variable and the frequency at which they run/update

Williangalvani, 2025-02-18:
It is hard to know if some configuration is missing withouth editing each of them:
[image]

(please close it if this ui is not on master 😅

ES-Alexander, 2025-02-18:
@Williangalvani can you submit this as a review to #1675 instead?

closed 2025-02-18 by ES-Alexander

rafaellehmkuhl, 2025-02-19:
Hey @Williangalvani now I got the point you we're mentioning. 

Any idea on how to squeeze that to this interface? Just make it larger? @ES-Alexander @ArturoManzoli any thoughts?

ArturoManzoli, 2025-02-19:
> Hey [@Williangalvani](https://github.com/Williangalvani) now I got the point you we're mentioning.
> 
> Any idea on how to squeeze that to this interface? Just make it larger? [@ES-Alexander](https://github.com/ES-Alexander) [@ArturoManzoli](https://github.com/ArturoManzoli) any thoughts?

We can highlight the link icon button with a slightly lighter color and also use an expansible row. With a click, another line is expanded below the current one, showing more detail about the action, such as linked variables, frequency etc.

I've done this to one variant of the the Settings manager. It was pretty nice, but will only be used with more nested configs. Probably on the near future.