bluerobotics/cockpit #1720: Dial: stop resizing value indicator with contents

ES-Alexander, 2025-02-27:
### Current behaviour

The value indicator resizes when the numbers change, which also wiggles the dial:

https://github.com/user-attachments/assets/b4c8f6d1-d773-4a1c-8380-7cf75803e7bb

### Expected or desired behaviour

It would be nice if the "value" indicator could be changed to a monospace font, and maybe just be set to a fixed size for the larger of the min/max value, so its width doesn't wiggle when you're changing the value.

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

ArturoManzoli, 2025-03-07:
Maybe this is an issue specific to Chrome on MacOS. Here on Linux, I can only see that wiggling when the number changes from 9999 to 10k. Lets ask @rafaellehmkuhl to test it on his mac.

https://github.com/user-attachments/assets/23de8fb1-2b44-4f47-9e8a-f4cd0cc86753

rafaellehmkuhl, 2025-03-12:
> Maybe this is an issue specific to Chrome on MacOS. Here on Linux, I can only see that wiggling when the number changes from 9999 to 10k. Lets ask [@rafaellehmkuhl](https://github.com/rafaellehmkuhl) to test it also on his mac.
> 

I think you're seeing the same problem that Eliot is mentioning. It happens on big numbers. If you use 10k or 9999 as max, the wiggle happens. 

For the 10k, it would need a bigger width or a "10k" string, while for the others, a monospace font could fix it.