bluerobotics/cockpit #1737: frontend: Mission Planning should support more mission commands

rmackay9, 2025-03-07:
### Current behaviour

Only "Pass-by" waypoint types are listed.  This is not a mission command that I'm familiar with but I think it refers to the WAYPOINT command.

[image]

### Expected or desired behaviour

ArduPilot Copter and Rover support many different mission commands that users would likely want to use.  For Copters at least the minimum set is listed below and below that is a more full list which is what Mission Planner supports.  I'm not necessarily saying we need all of these but certainly a few more would be good

Takeoff
Waypoint
Land
Return-to-launch

[image]

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-03-07:
Thanks for the report Randy!

I thought we already had an issue to track that, but we didn't. Added it to the Boat milestone as well.

We do have a wip branch around that feature. We plan to get back working on it in the next quarter.

ES-Alexander, 2025-08-06:
The high priority aspect of this is initial support, which is fine to be reasonably bare-bones - it just needs to be possible to specify different commands at waypoints, even if it's somewhat cumbersome to do so (e.g. manually specifying names and field values).

The nicer aspects of the interface (e.g. auto-populating fields and descriptions, and adding visual tools to help with things like setting camera angles, and icons on the waypoints to see at a glance which ones do what) can be discussed in a follow-on Issue and implemented later.

PR #2054 (2025-09-17): Add initial support for command waypoints
Allows someone to choose one of their waypoints and change it from "Pass by" to "MAVLink Generic Waypoint". When this is done 5 new fields appear to allow the user to explicitly set the "MAV_CMD" as well as the 4 parameters.

Conversion from the vehicle to Cockpit also takes those into account (any waypoint that is not a `MAV_CMD_NAV_WAYPOINT` is casted to a "MAVLink Generic Waypoint".

In the video it can be seen that two waypoints were modified to use a MAV_CMD_DO_JUMP and skip waypoints 4 and 7 from the mission, going directly to 5 and 2, respective [... cut]
closed 2025-09-17 by rafaellehmkuhl