bluerobotics/cockpit #1759: Refactor MANUAL_CONTROL support to be directly configurable

ES-Alexander, 2025-03-18:
### Current behaviour

Joystick `MANUAL_CONTROL` support has opaque internals (which can make transferring between control station software options needlessly ambiguous), and cannot make full use of Cockpit's Data Lake or Actions systems.

### Expected or desired behaviour

`MANUAL_CONTROL` support could be exposed as a default Cockpit Action based on (and triggered by?) a set of predefined compound Data Lake variables (`Axis X`/`Y`/`Z`/`R`/`S`/`T`, `buttons`, `buttons2`, etc)[^1], which could in turn be composed of values that get set nicely using the existing streamlined interface, but could also be modified manually to include other Data Lake variables (or combinations) as relevant (e.g. to [allow two spring-loaded triggers to be combined as positive/negative throttle](https://discuss.bluerobotics.com/t/how-to-make-a-custom-controller-preset-for-rov-controlling/19713)).

Having a single Action, with the standard configuration for frequency limits, and the ability to disable it, would form a decent basis for implementing #1754.

[^1]: building on #1640

### Additional information

Idea raised in [this comment](https://github.com/bluerobotics/cockpit/issues/843#issuecomment-2733050575)

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

AbnerZ74, 2025-03-18:
Technically there is no way of making a true custom control, but a semi-custom control is surly acheivable

ES-Alexander, 2025-05-12:
#1882 and #1883 could help with the implementation of this, although it should be minimally possible to achieve once #1658 is implemented, just somewhat awkward with some of the variable mappings, especially since there'd be no explicit link between the assigned data-lake variables and the corresponding autopilot button functions (plus the redirection that the mapping of data-lake variables to MAVLink MANUAL_CONTROL button positions would be in a compound variable, and the fact that assigning a currently unused button function type would require separately setting a parameter in the autopilot).

It could be helpful to auto-generate data-lake variables for the names of configured autopilot button functions (e.g. `mmc-lights1-cycle` / `mmc-relay4-toggle`), and provide some kind of helper tool for normal joystick configuration that handles automatically adding/removing those variables from the bitwise OR of the compound button variable(s), as well as adding/replacing the relevant autopilot parameters if a function is assigned that isn't already set to a parameter.