bluerobotics/cockpit #1780: frontend: standalone video folder should be moved to /Documents directory

vshie, 2025-03-26:
### Current behaviour

Many users will struggle to find the Cockpit video folder in Standalone, we've seen this in the inbox already! By moving to Documents folder, we match QgroundControl paradigm, which will help with switchover. 

### Expected or desired behaviour

/User/Documents/Cockpit/Video instead of /User/Cockpit/Video please!

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2025-12-05:
There could be some unwanted behaviour if we make it move old files when someone sets a new location (e.g. if they change the location to or from an external/networked drive, and don't want the old files to be removed from the old drive).

We could add a button to "transfer old files" that's only visible in the session where they've changed it, but functionality like that is also code that needs writing and maintaining, so might make the feature take longer 🤷‍♂️

vshie, 2025-12-05:
gotcha. Well, I'd be ok with it notifying user they can change it, and if they do, it not necessarily moving files, just starting over in new location? less complexity?