bluerobotics/cockpit #1791: Telemetry file not being processed for some variables

ArturoManzoli, 2025-04-01:
### Bug description

When including a variable such as HOME_POSITION/x, being displayed on a VGI, the telemetry file generation crashes, generating a broken second video on the library, also making the main video thumb not available.

[image]

[image]

### Steps to reproduce

- Create a VGI with the variable HOME_POSITION/x
- Add it to the on-screen telemetry layer
- Record a video
- Check the video library

### Primary pain point(s)

- This may be one of multiple variables breaking the video and telemetry processing.
- Users won't have a clue why the video processing is weird.

We should make the telemetry generation process more resilient to unpredicted variable contents;

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-04-02:
Humn, this is a important bug. We should fix it soon.

ArturoManzoli, 2025-04-08:
Testing with a new setup made it possible to confirm that the issue was a false positive caused by a stream duplication problem. Subtitles are being generated normally using the same variable mentioned in the bug description, and no related issues have been observed.

[image]

ArturoManzoli, 2025-04-08:
Closing the issue as false positive

closed 2025-04-08 by rafaellehmkuhl