bluerobotics/cockpit #1792: frontend: camera and gimbal control screen

rmackay9, 2025-04-02:
### Current behaviour

It would be great if we could get some improved camera and gimbal controls to help users with live video.

The [QGC controls](https://ardupilot.org/copter/docs/common-mount-targeting.html#qgc-gimbal-controls) might be the best guide but in short these are the controls that most people would like:

1. click on the live video to change the camera gimbal's pitch/yaw angle
2. zoom in/out control
3. auto focus button (optional)
4. take picture button
5. record start/stop button

The ardupilot dev site has a couple of pages which could help figuring out the mavlink messages to use:

- gimbal controls: https://ardupilot.org/dev/docs/mavlink-gimbal-mount.html
- camera controls: https://ardupilot.org/dev/docs/mavlink-camera.html

### Expected or desired behaviour

users would appreciate being able to easily control a camera gimbal

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-04-02:
Hey Randy!

It would be cool to have this feature!

We currently have a big backlog of work to finish full support for the ROV and Boat, so it will be difficult to work on this anytime soon, but if you happen to know about anyone who would be interested in working on this, please let us know! We can definitely help with some guidance in the internals of Cockpit.

rafaellehmkuhl, 2026-10-26:
Prioritizing this one.