bluerobotics/cockpit #1798: All joystick mapping options should be under the same list

rafaellehmkuhl, 2025-04-03:
Asking the user which "type" of map option they want (e.g.: mavlink-manual-control, cockpit-actions, data-lake-variables) is unintuitive.

We can name the types under the options with something like a tag, so the information is there.

ES-Alexander, 2025-04-23:
#1755 is relevant.

rafaellehmkuhl, 2025-04-23:
#1755 is actually being addressed on #1836 as well!

I agree on the grouping idea. It's something we should put some thoughts/time on. I don't like that "Camera Focus Decrease" is under "Data Lake Variables", as it doesn't say much to a regular user.

rafaellehmkuhl, 2025-04-23:
> [#1755](https://github.com/bluerobotics/cockpit/issues/1755) is relevant (i.e. icons would help make a single list more viable/intuitive).
> 
> We've also had internal discussions that ideally functions would be grouped by feature, e.g. things related to vehicle motion, peripherals, interface controls, etc).

#1755 is actually being addressed on #1836 as well!

I agree on the grouping idea. It's something we should put some thoughts/time on. I don't like that "Camera Focus Decrease" is under "Data Lake Variables", as it doesn't say much to a regular user.

PR #1836 (2025-04-24): Simplify joystick mapping list
This PR simplifies a lost the list of available joystick mapping options. To do that it:
- Put all options under the same list, making it clear from where are they from
- Always show the search input
- Removes `MANUAL_CONTROL` options that are also available as Cockpit Actions
- Removes MAVLink Zoom and Focus actions, which should only be triggered automatically (from changes in the value of the increase/decrease variables)

https://github.com/user-attachments/assets/bc6f8791-99d5-4654-9226-c1906fb4b2eb

Fix #1755 
Fix #1798
Fix #1799 

To be merged after #1835.
closed 2025-04-24 by rafaellehmkuhl