bluerobotics/cockpit #1799: We should remove "duplicated" joystick options

rafaellehmkuhl, 2025-04-03:
For example, today we have both the `Cockpit Action - Arm` and the `MAVLink Manual Control - Arm` options. Which one should the user choose?

They basically do the same thing, under different approaches.

We should hide the MAVLink Manual Control options that have counterparts (like the arm and disarm) by default, and only show them if the user asks for it (a checkbox somewhere).

ES-Alexander, 2025-04-23:
Note that they also need to be shown when they're currently selected on the autopilot, even if the general selector for _changing_ mappings is hiding them as alternative options.

rafaellehmkuhl, 2025-04-24:
> Note that they also need to be shown when they're currently selected on the autopilot, even if the general selector for _changing_ mappings is hiding them as alternative options.

On #1836 I'm only hiding them as options on the dialog for changing the input mapping. They still appear normally on the visual and table pages.

PR #1836 (2025-04-24): Simplify joystick mapping list
This PR simplifies a lost the list of available joystick mapping options. To do that it:
- Put all options under the same list, making it clear from where are they from
- Always show the search input
- Removes `MANUAL_CONTROL` options that are also available as Cockpit Actions
- Removes MAVLink Zoom and Focus actions, which should only be triggered automatically (from changes in the value of the increase/decrease variables)

https://github.com/user-attachments/assets/bc6f8791-99d5-4654-9226-c1906fb4b2eb

Fix #1755 
Fix #1798
Fix #1799 

To be merged after #1835.
closed 2025-04-24 by rafaellehmkuhl