bluerobotics/cockpit #1809: Mavlink messages from other components/systems can not be used

fseegraeber, 2025-04-08:
Hi there,

I've encountered following problem:

Currently all mavlink messages with `system_id  != 1` or `component_id != 1` are discarded:
https://github.com/bluerobotics/cockpit/blob/83ebc637923eeba97de3afe900ebc91902d884b1/src/libs/vehicle/ardupilot/ardupilot.ts#L309-L311

This is a problem when trying to use mavlink messages from a separate component e.g. in a Very Generic Indicator.  If you have control over the communicating component you can work around this by setting system and component id of its messages to 1. Although with certain side effects, see below. It would be nice if this was not necessary.

As a concrete example I've recently integrated a separate hardware component into a BlueROV system. The components sends a sensor reading via mavlink to be displayed in Cockpit and also listens to a mavlink message to control attached lights. Using the above workaround you can get the messages to show up in Cockpit, with one caveat: The mavlink router service will stop forwarding messages of a certain component id to an endpoint if that endpoint sends messages containing said id. This means that in this concrete example you need to configure two separate endpoints for incoming and outgoing messages because the outgoing messages with component_id 1 will prevent all other BlueROV messages from being forwarded to it.

rafaellehmkuhl, 2025-04-08:
Hey @fseegraeber!

Although links to control multiple vehicles (natively) is still far on our timeline, since it involves a good amount of work, receiving telemetry from other vehicles or components is definitely something we want to support, specially for using on VGIs and Plotters.

I will investigate it and try to come up with a quick solution. Will bring news about it during the next days.

rafaellehmkuhl, 2025-04-10:
@fseegraeber can you test #1815 and see if it fixes the problem?

closed 2025-11-12 by rafaellehmkuhl