bluerobotics/cockpit #1824: The "tie-joystick-mapping-to-vehicle-type" feature is very much not clear

rafaellehmkuhl, 2025-04-17:
One looking today at the page below probably has no idea what much of those buttons do. Most people probably just ignore everything and go directly to the buttons mapping.

[image]

The feature to tie joystick profiles to vehicle types specially, should be on a separated section, so it is clear to the user what is happening, probably with arrows connecting all the mapped options (MAV_TYPE_SUBMARINE -> "ROV functions mapping", etc).

ES-Alexander, 2025-04-23:
#1547 may be a preferable solution.

rafaellehmkuhl, 2025-04-23:
> [#1547](https://github.com/bluerobotics/cockpit/issues/1547) may be a preferable solution.

I'm not totally convinced that #1547 is the right approach, but the two implementations are are very far in terms of amount of work invested. One is a simple UI change, while the other involves migrations and changes in core structures.

I think it would be valuable to address this in a simpler way, at least till we have the time to decide and invest on the long term solution.