bluerobotics/cockpit #1843: frontend:  Actions at waypoint equivalency to QGC

vshie, 2025-04-24:
### Current behaviour

QGround Control allows you to set various actions at a waypoint, converting the waypoint to an action at that step. The most commonly used of these is "jump to" where you can enter in any waypoint number, and when the mission reaches this point it will loop back to a previously reached waypoint. The only configuration parameter provided is # of repetitions, which allows you to drive a mission n number of times. 

### Expected or desired behaviour

Match some, if not all waypoint actions that QGround Control provides. 

[image]

[image]

[image]

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-04-24:
Nice. This should be on of the first when we start the Boat equivalency.

Williangalvani, 2025-08-05:
I'm pretty sure these are all handled at autopilot level. It should be a matter of just allowing to pick different entries and filing the parameters manually (I'd rather not have the params names than miss the feature altogether)

ES-Alexander, 2026-03-19:
Partially solved by #2054, but the interface is quite minimal at the moment (all parameters get specified, instead of only the necessary ones, and there aren't nice names/descriptions of the relevant fields).