bluerobotics/cockpit #1844: Going to vehicle setup, pwm output page with one client with cockpit open  on another causes control input loss in cockpit session

vshie, 2025-04-24:
### Bug description

Not tested if this happens with a single client. 
If vehicle setup page is open on one client in BlueOS (specifically PWM outputs, 1.4 stable), then motor test function seems to cause cockpit input loss - arm/disarm state does not change. 

### Steps to reproduce

Open cockpit on one computer. Open BlueOS on another. Verify control in cockpit (e.g. camera tilt servo.) Open vehicle setup > pwm outputs. Verify control of vehicle lost on first computer - cannot adjust camera tilt, joysticks do not cause motor motion even if system armed. arm/disarm butons have no effect. 

### Primary pain point(s)

Check on what a pwm channel is set to, and outputting, lose vehicle control....

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-04-24:
@Williangalvani do you have any clue why that could happen?

rafaellehmkuhl, 2025-04-28:
> Yes this is a BlueOS issue. it sends commands if the vehicle is armed (and IIRC it also needs to be on manual mode) when on that page, regardless of who armed it. 

Ok. Will keep the issue open here for us to track its fix as well.