bluerobotics/cockpit #1856: frontend: request data from autopilot

rmackay9, 2025-04-30:
### Current behaviour

If a user connects Cockpit to a Copter they will not see the vehicle on the map because Copter (unlike Plane and Rover) does not proactively send out many mavlink messages on all mavlink channels

In case it matters, this was originally done this way to reduce CPU overhead back in the days when AP was only running on the low end hardware.  The AP flight code team has considered changing this to sent at 1hz by default.

In any case, the user probably needs to be able to set the data rates because telemetry bandwidth varies greatly amongst different setups.  The AP developer wiki has this page explaining [how to request data from the autopilot](https://ardupilot.org/dev/docs/mavlink-requesting-data.html).  Maybe we can use the simplest REQUEST_DATA_STREAM method but most ground stations provide a way to set the rate of the streams.  Some (like mission planner) also include a method to set the rate of individual messages (aka MESSAGE_INTERVAL)

### Expected or desired behaviour

Some data should be streamed by default, users should also be able to change the default rate

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-04-30:
We actually do that already 🧐

I will take a look to see if it could be that there's something wrong on that end.

rafaellehmkuhl, 2025-04-30:
About being able to define the value for each message, will make it happen, but about them not being requested by default today for you, I'm curious on what could be happening. Mine is working fine, and I don't see any clear caveats in the implementation. This is what Cockpit is sending here:

```json
{
  "message": {
    "command": {
      "type": "MAV_CMD_SET_MESSAGE_INTERVAL"
    },
    "confirmation": 0,
    "param1": 22,
    "param2": 500000,
    "param3": 0,
    "param4": 0,
    "param5": 0,
    "param6": 0,
    "param7": 0,
    "target_component": 1,
    "target_system": 1,
    "type": "COMMAND_LONG"
  },
  "status": {
    "time": {
      "counter": 83398,
      "first_update": "2025-04-28T12:05:06.677196465Z",
      "frequency": 0.4781830906867981,
      "last_update": "2025-04-30T12:31:52.678844212Z"
    }
  }
}

```

It does that for all messages in [this list](https://github.com/bluerobotics/cockpit/blob/master/src/libs/vehicle/ardupilot/defaults.ts), and the request is called [in the vehicle constructor already](https://github.com/bluerobotics/cockpit/blob/648e518208aa97fb9757227396151b4168f3a21c/src/libs/vehicle/ardupilot/ardupilot.ts#L128).

rafaellehmkuhl, 2025-04-30:
@rmackay9 while we better understand the bug you're seeing, I've implemented the option for the user to set their own message rate individually on #1858. Let me know if you see something missing.

rmackay9, 2025-05-01:
Hi @rafaellehmkuhl,

Thanks for the PR (I plan to test it soon).  I wonder which vehicle type is installed on the autopilot you're using?  It needs to be Copter to see the issue I think

rmackay9, 2025-05-01:
I'll go ahead and close this, I definitely saw the issue of course, but perhaps I was still messing around with the configuration at the time.  Upon restarting everything just now and re-testing it all worked fine.

rafaellehmkuhl, 2025-05-01:
@rmackay9 I'm happy that it's working now! If you see the problem again, feel free to reopen the issue.

closed 2026-01-16 by rafaellehmkuhl