bluerobotics/cockpit #1868: frontend: combine button-based mini-widgets

ES-Alexander, 2025-05-06:
### Current behaviour

There's a growing list of mini-widgets that are just a single button (fullscreen, edit mode, joystick connection indicator, takeoff, change alt), with an icon (possibly indicating a state) and/or some text. Maintaining and documenting these separately is annoying, and it makes the widgets list needlessly long, which makes it harder to find desired widgets.

### Expected or desired behaviour

Combining widgets with similar functionality/interfaces allows users to more easily reason about the application, while also reducing the maintenance load to keep interfaces consistent.

1. Similar to the `VeryGenericIndicator`, I think the button widgets should be combined into a set of presets for the Action Button widget, and allow specifying the button icon/colour depending on the state of a data-lake variable.
2. We could potentially even add Action functionality to the `VeryGenericIndicator` and turn it into a more generic `ViewDo` widget. This would combine a bunch of display logic, while also making it easier for existing indicator widgets to have some logical function associated with them (e.g. centring the camera, toggling input hold, toggling the lights, etc).

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2025-05-06:
Totally!

ES-Alexander, 2025-11-21:
Running with the `ViewDo` idea, it should have the following configuration options:
- name (optionally displayed)
- display elements (choose at least 1)
    - label (optional)
    - icon(default)/button/switch/checkbox/slider/dial (optional)
    - value entry (optional)
    - value units (optional)
    - value inc/dec buttons (optional)
    - base colour (button face, slider, dial, checkbox+switch OFF)
    - accent colour (text, dial's knob indicator, checkbox+switch ON)
- display source (optional)
    - a data-lake variable (with a tree selector?)
    - ideally this would have an option to filter by the type of value units (e.g. "distances" / "temperature" / etc)
    - scale factor (if unknown input units)
- action variable (optional)
    - matches the input variable by default, if it is writeable, else none
    - min val / max val (default 0 / 1)
- triggers (optional)
    - growable list of trigger threshold(s), with corresponding actions, display colours, and display icons
    - default triggers could be set up at 

From there we could represent several of our existing widgets as presets, and replace the input-widget set entirely...